makerdiary/nrf52840-mdk #1: Bill of Materials

danieldeesousa, 2018-09-10:
I was wondering if there was a BOM for the hardware for this project in case I wanted to base my hardware project off of this sometime in the future.

caizelin, 2018-09-11:
@danieldeesousa Hi, you can check the [schematic](https://github.com/makerdiary/nrf52840-mdk/blob/master/docs/hardware/nrf52840-mdk-schematic_v1_0.pdf) for more hardware details and choose the right components for your own project.

saulpw, 2018-10-18:
Seconded.  In particular wanting to know the external flash part number.  I am giving myself a headache trying to read it off the board and was hoping that it was mentioned somewhere.

caizelin, 2018-10-19:
> Seconded. In particular wanting to know the external flash part number. I am giving myself a headache trying to read it off the board and was hoping that it was mentioned somewhere.

@saulpw The PN of external flash is MX25R6435F, the same as Nordic's nRF52840-DK. You can use the QSPI example from nRF5 SDK by adapting the pin number.

caizelin, 2019-06-21:
Pls reopen this issue if you have any concerns. Thanks!

closed 2019-06-21 by caizelin

g-e-r, 2019-08-05:
I’m interested in knowing the regulator parts.