makerdiary/nrf52840-mdk #2: Programming

danieldeesousa, 2018-09-13:
Can I program this board over an interface which is not USB such as SWD?

If so, what would you recommend as a debugger?

Lindenk, 2018-09-27:
It looks like there are pins for SWD/JTAG on the right of the board. I'm currently using [this programmer](https://www.adafruit.com/product/3571) and you probably will also need a break out board [like this](https://www.adafruit.com/product/2743).

wez, 2018-09-27:
I hooked mine up to a jlink using the adafruit breakout board.  Pinout and some more info is captured here: https://github.com/nrf-rs/nrf52840-mdk-rs#jlink

caizelin, 2018-10-04:
@danieldeesousa The SWD pins are on the right header of the board. The board has a [DAPLink](https://github.com/ARMmbed/DAPLink) debugger integrated. It should be more convenient. If you want to use an external debugger, J-Link is recommended.

caizelin, 2019-06-21:
Pls reopen this issue if you have any concerns. Thanks!

closed 2019-06-21 by caizelin