makerdiary/nrf52840-mdk #4: Antenna selection

der-tobi, 2018-10-15:
According to the Datasheet we have a selectable U.FL Connector for an external Antenna on the nRF52840-MDK.
How can we select between the internal (Chip) an the external Antenna?
Do you have a code-example of the usage for the nRF-SDK 15.2?
I just need a BLE-Beacon every x ms over the external Antenna.

(https://community.makerdiary.com/t/u-fl-connector-selection/212)

caizelin, 2018-10-16:
Hi there,
A rework must be performed before the U.FL connector can be used; this involves swapping the position of a 3.3nH chip inductor. The modified board would appear as follows.
* **Here is for the external antenna:**
[image]

* **And here is for the chip antenna:**
[image]

nRF-SDK 15.2 support is working in process. I will update the examples later. 

Best,
Zelin

der-tobi, 2018-10-21:
Oops, that kind of "selectable". 
Thank you for your answer.
Kind regards,
Tobi

closed 2018-10-21 by der-tobi

aeozyalcin, 2020-07-21:
@caizelin sorry to bring up an old thread, but can you confirm the 1 pF capacitor does not need to be moved? Looking at the schematic, the chip antenna uses L4 (3.3nH) and C38 (1pF) as a matching network for 50 ohm chip antenna. The instructions here only say to move L4 (3.3 nH). Can you confirm C38 (1pF) does **_not_** need to be moved the empty C46 footprint? I figured we would need to move both to match 50 ohm impedance. 

Thanks!

cdeletre, 2021-05-31:
I've manage to make the external antenna working by only moving the L4. That wasn't a piece of cake with this size of component (0402) but I did it !