makerdiary/nrf52840-mdk #5: External U.FL connector selector

SethSenpai, 2018-10-23:
Hi there,

I was wondering how to select the external antenna connector on the board. I could not find anything in the documentation regarding this. I presume I need to re-solder a capacitor. If you so kindly could indicate which one I need to move that would be great!

Kind regards,
Seth

SethSenpai, 2018-10-23:
Nevermind I just checked the closed issues and the explanation is in there.

closed 2018-10-23 by SethSenpai

SethSenpai, 2018-10-23:
Perhaps it is prudent to provide this documentation on the website as well. Digging through closed github issues is not everyones cup of tea.

caizelin, 2018-10-23:
> Perhaps it is prudent to provide this documentation on the website as well. Digging through closed github issues is not everyones cup of tea.

You are right. I will add this to the wiki.