makerdiary/nrf52840-mdk #6: Examples are not working with latest nRF5_SDK_15.2.0_9412b96

dhruvkakadiya, 2018-11-24:
There are **experimental_*** libraries(e.g. experimental_log, experimental_memobj, experimental_ringbuf etc.) in older SDK which got updated in the latest release so Makefiles are not working with the latest SDK.

caizelin, 2018-12-05:
> There are **experimental_*** libraries(e.g. experimental_log, experimental_memobj, experimental_ringbuf etc.) in older SDK which got updated in the latest release so Makefiles are not working with the latest SDK.

Hi there,
The latest nRF5_SDK_15.2.0_9412b96 is supported now.  Just pull and play! 🎉

dhruvkakadiya, 2018-12-05:
Thank you, @caizelin. I will test out few examples and if everything looks good then I'll close this ticket.

With older SDK `nRF5_SDK_15.0.0_a53641a`, I've been playing to get working on-chip flash_fds example(FDS - Flash Device Storage) and so far I didn't get success for it. I was able to write a config file to the file system but then while trying to update that file ran into some issues and then controller keeps on resetting itself. Since you've now added latest SDK support so I'll try again that code with it and see if it's working now.

Were you able to get that example working because I don't see that example inside `examples/nrf5-sdk` directory as officially supported yet?

caizelin, 2019-06-21:
Pls reopen this issue if you have any concerns. Thanks!

closed 2019-06-21 by caizelin