makerdiary/nrf52840-mdk #10: No partition shown removable drive

aleek, 2018-12-29:
Hi!

Gentoo Linux here! When I plugged my shiny new nrf52840-mdk, both serial and drive shown, but no partition there. I believe I should make one, along with partition table. Can you walk me through the process?

```
[ 2158.907931] usb 1-3.4: new high-speed USB device number 10 using ehci-pci
[ 2158.998448] usb 1-3.4: New USB device found, idVendor=0424, idProduct=2422, bcdDevice= 0.a0
[ 2158.998450] usb 1-3.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2158.998710] hub 1-3.4:1.0: USB hub found
[ 2158.998815] hub 1-3.4:1.0: 2 ports detected
[ 2160.451918] usb 1-3.4.1: new full-speed USB device number 11 using ehci-pci
[ 2160.548241] usb 1-3.4.1: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[ 2160.548243] usb 1-3.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2160.548244] usb 1-3.4.1: Product: DAPLink CMSIS-DAP
[ 2160.548245] usb 1-3.4.1: Manufacturer: ARM
[ 2160.548246] usb 1-3.4.1: SerialNumber: 1026000003f3816000000000000000000000000097969902
[ 2160.548882] usb-storage 1-3.4.1:1.0: USB Mass Storage device detected
[ 2160.549012] scsi host7: usb-storage 1-3.4.1:1.0
[ 2160.551159] hid-generic 0003:0D28:0204.0008: hiddev98,hidraw7: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-0000:00:12.2-3.4.1/input3
[ 2160.552168] cdc_acm 1-3.4.1:1.1: ttyACM0: USB ACM device
[ 2161.567632] scsi 7:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
[ 2161.567824] sd 7:0:0:0: Attached scsi generic sg5 type 0
[ 2161.568366] sd 7:0:0:0: [sdf] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
[ 2161.568988] sd 7:0:0:0: [sdf] Write Protect is off
[ 2161.568990] sd 7:0:0:0: [sdf] Mode Sense: 03 00 00 00
[ 2161.569614] sd 7:0:0:0: [sdf] No Caching mode page found
[ 2161.569616] sd 7:0:0:0: [sdf] Assuming drive cache: write through
[ 2161.595692] sd 7:0:0:0: [sdf] Attached SCSI removable disk
```

```
$  fdisk -l /dev/sdf                                                                                                                                                                                                                                                                    
Disk /dev/sdf: 64.1 MiB, 67174400 bytes, 131200 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```

caizelin, 2019-01-01:
@aleek The USB Mass Storage device is detected. Could you program nRF52840 chip by copying the firmware to DAPLink disk?

aleek, 2019-01-01:
ok, it was so simple...
```
sudo mount /dev/sdf /mnt/usb 
```
I thought there should be some partition table.

closed 2019-01-01 by aleek