makerdiary/nrf52840-mdk #11: Unable to use make flash

bsingh5266, 2018-12-29:
`**Singh:armgcc dhillon$ make flash**
DONE nrf52840_xxaa
Flashing: _build/nrf52840_xxaa.hex
pyocd-flashtool -t nrf52 -se _build/nrf52840_xxaa.hex
Traceback (most recent call last):
  File "/usr/local/bin/pyocd-flashtool", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3095, in <module>
    @_call_aside
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3081, in _call_aside
    f(*args, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3108, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 658, in _build_master
    ws.require(__requires__)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 959, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 846, in resolve
    raise DistributionNotFound(req, requirers)
**pkg_resources.DistributionNotFound: The 'enum34' distribution was not found and is required by pyocd**
make: *** [flash] Error 1
Singh:armgcc dhillon$ `

caizelin, 2019-01-01:
@bsingh5266 Could you check the version of `pyocd-flashtool` or reinstall `pyocd`?
Here is the command: `pyocd-flashtool --version`.

caizelin, 2019-06-21:
Pls reopen this issue if you have any concerns. Thanks!

closed 2019-06-21 by caizelin