makerdiary/nrf52840-mdk #12: DAPLINK programming issues and keeps disconnecting

lukaszmargielewski, 2019-01-27:
I m using Mac OS and pyocd tool to program mdk board. I have following issues since the beginging: 

1. Board keeps disconnecting with message: "Disk Not Ejected Properly Eject DAPLINK before disconnecting or turning off" 

2. Programming board works only for few times and then keeps failing more and more until I restart Mac for it to work for a while again. 
I am using following programming command:
`pyocd-flashtool -t nrf52 -ce binay.hex`
I am also using screen command to listen to logs from the board:
`screen /dev/tty.usbmodem141102 115200`

Upgrading firmware to the most recent version did not help.

To be honest I am not sure what is the source of the problem (hardware / software / Mac OS) and if it is the correct place but I couldn't find solution myself nor better place to ask for help.

caizelin, 2019-01-27:
> I m using Mac OS and pyocd tool to program mdk board. I have following issues since the beginging:
>
> 1. Board keeps disconnecting with message: "Disk Not Ejected Properly Eject DAPLINK before [...]

Could you have a try on another host PC to check if it is caused by the host?

stefanosky, 2019-02-01:
hi
same situation to me, in a new mac os.
maybe the current of the circuits leds? it's too much for mac book pro?

stefanosky, 2019-02-02:
still there the problem of disconnected and connected device
on mac book pro 2018 with original adapter usb c -> usb

caizelin, 2019-02-02:
@lukaszmargielewski `brew install libusb` should be implemented. Also see #14

stefanosky, 2019-02-02:
whoaw what a mess, with the official apple usb adapter didn't works, but with a chinese adapter for android phone usbc -> usb works, the connection is stable, and resolve this issue of disconnected continuosly

caizelin, 2019-02-02:
> whoaw what a mess, with the official apple usb adapter didn't works, but with a chinese adapter for android phone usbc -> usb works, the connection is stable, and resolve this issue of disconnected continuosly

You can also connect the board without an adapter, if you have a USB-C to USB-C cable. I have tested that it works on a new MBP with USB-C interface.

lukaszmargielewski, 2019-02-06:
@caizelin : Sorry for late response - it took me a while to find other Mac and install all tools. Similar things there. I also tried uninstalling pyocd and sung dappling only but the same disconnections happen. I see other people having issues with usb but my MacBook is having standard USB ports.

caizelin, 2019-06-21:
Pls reopen this issue if you have any concerns. Thanks!

closed 2019-06-21 by caizelin