makerdiary/nrf52840-mdk #14: after make flash the code is not uploaded on board

stefanosky, 2019-02-01:
on a macos
installed pip and brewer, pyocd and gcc toolkit

after command make flash the code is not uploaded on board
with the blinky led example works, then i try with ble blinky and not works

thanks

`MacBook-Pro-di-Stefano:armgcc stefano$ make flash
Assembling file: gcc_startup_nrf52840.S
Compiling file: nrf_log_backend_rtt.c
Compiling file: nrf_log_backend_serial.c
Compiling file: nrf_log_backend_uart.c
Compiling file: nrf_log_default_backends.c
Compiling file: nrf_log_frontend.c
Compiling file: nrf_log_str_formatter.c
Compiling file: app_button.c
Compiling file: app_error.c
Compiling file: app_error_handler_gcc.c
Compiling file: app_error_weak.c
Compiling file: app_scheduler.c
Compiling file: app_timer.c
Compiling file: app_util_platform.c
Compiling file: hardfault_implementation.c
Compiling file: nrf_assert.c
Compiling file: nrf_atfifo.c
Compiling file: nrf_atflags.c
Compiling file: nrf_atomic.c
Compiling file: nrf_balloc.c
Compiling file: nrf_fprintf.c
Compiling file: nrf_fprintf_format.c
Compiling file: nrf_memobj.c
Compiling file: nrf_pwr_mgmt.c
Compiling file: nrf_ringbuf.c
Compiling file: nrf_section_iter.c
Compiling file: nrf_strerror.c
Compiling file: system_nrf52840.c
Compiling file: boards.c
Compiling file: nrf_drv_clock.c
Compiling file: nrf_drv_uart.c
Compiling file: nrfx_clock.c
Compiling file: nrfx_gpiote.c
Compiling file: nrfx_power_clock.c
Compiling file: nrfx_prs.c
Compiling file: nrfx_uart.c
Compiling file: nrfx_uarte.c
Compiling file: main.c
Compiling file: SEGGER_RTT.c
Compiling file: SEGGER_RTT_Syscalls_GCC.c
Compiling file: SEGGER_RTT_printf.c
Compiling file: ble_advdata.c
Compiling file: ble_conn_params.c
Compiling file: ble_conn_state.c
Compiling file: ble_srv_common.c
Compiling file: nrf_ble_gatt.c
Compiling file: nrf_ble_qwr.c
Compiling file: utf.c
Compiling file: ble_lbs.c
Compiling file: nrf_sdh.c
Compiling file: nrf_sdh_ble.c
Compiling file: nrf_sdh_soc.c
Linking target: _build/nrf52840_xxaa.out
   text	   data	    bss	    dec	    hex	filename
  26992	    164	   2488	  29644	   73cc	_build/nrf52840_xxaa.out
Preparing: _build/nrf52840_xxaa.hex
Preparing: _build/nrf52840_xxaa.bin
DONE nrf52840_xxaa
Flashing: _build/nrf52840_xxaa.hex
pyocd-flashtool -t nrf52 -se _build/nrf52840_xxaa.hex
WARNING:root:pyocd-flashtool is deprecated; please use the new combined pyocd tool.
INFO:board:Target type is nrf52
INFO:root:DP IDR = 0x2ba01477
INFO:root:AP#0 IDR = 0x24770011
INFO:root:AP#1 IDR = 0x02880000
INFO:root:AP#0 ROM table #0 @ 0xe00ff000 (designer=244 part=008)
INFO:root:[0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
INFO:root:[1]<e0001000:DWT class=14 designer=43b part=002>
INFO:root:[2]<e0002000:FPB class=14 designer=43b part=003>
INFO:root:[3]<e0000000:ITM class=14 designer=43b part=001>
INFO:root:[4]<e0040000:TPIU-M4 class=9 designer=43b part=9a1 devtype=11 archid=0000 devid=0:0:ca1>
INFO:root:[5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
INFO:root:CPU core is Cortex-M4 r0p1
INFO:root:FPU present: FPv4-SP
INFO:root:4 hardware watchpoints
INFO:root:6 hardware breakpoints, 4 literal comparators
[====================] 100%
INFO:pyocd.flash.loader:Programmed 27156 bytes (7 pages) at 12.19 kB/s (7 pages unchanged)
#pyocd-flashtool -t nrf52 -se _build/nrf52840_xxaa.hex
MacBook-Pro-di-Stefano:armgcc stefano$ 
`

caizelin, 2019-02-02:
@stefanosky the ble blinky example needs the SoftDevice, so before `make flash` command, you should download the SoftDevice by using `make flash_softdevice` command.

stefanosky, 2019-02-02:
Ok thanks, can you post here the line of code to put in makefile.c  to resolve flash_softdevice? Because is missing, both flash_all and flash_device.

I think on macos there is some issue about current voltage stability, the dap drive is loaded and ejected several times and the times that the code was not loaded it's due to not sync with board. You tested with macbook pro 2018?

Thanks

caizelin, 2019-02-02:
Here is the ble_app_blinky example which contains `flash_softdevice`: https://github.com/makerdiary/nrf52840-mdk/tree/master/examples/nrf5-sdk/ble_app_blinky

My host is macbook pro 2014. Have you installed `libusb`? Try `brew install libusb`. I will find a new MBP to test.
Thanks for your feedback!

stefanosky, 2019-02-02:
Another thing minor, pyocd warning to use new command instead of deprecated pyocd-flashtool. Like "pyocd flash ... "
But i don't know how to rewrite. I tried but something goes wrong

caizelin, 2019-02-02:
> Another thing minor, pyocd warning to use new command instead of deprecated pyocd-flashtool. Like "pyocd flash ... "
> But i don't know how to rewrite. I tried but something goes wrong

Could you post the output log? Although `pyocd-flashtool` is deprecated, it still works currently. 
Anyway, you can use `pyocd` like this to flash:
```
pyocd flash -t nrf52 -e sector <your hex file>
```
Just type `pyocd flash -h` to get the help info.

stefanosky, 2019-02-02:
> Here is the ble_app_blinky example which contains `flash_softdevice`: https://github.com/makerdiary/nrf52840-mdk/tree/master/examples/nrf5-sdk/ble_app_blinky
>
> My host is macbook pro 2014. Have you [...]

I reinstall libusb now i flashed with softdevice succesfully.
i found a guide for nrf52 on mac and i missed to add some path, i will do later
for other people interested, i found this one
https://aaroneiche.com/2016/06/01/programming-an-nrf52-on-a-mac/

thanks

caizelin, 2019-02-02:
@stefanosky Awesome! 👍

stefanosky, 2019-02-02:
i updated also with the new command pyocd flash, works like a charm!
thanks!
now i have to study how to do Thread networks

stefanosky, 2019-02-02:
now i use the board with usbc female, i'm looking to shrink project board, and the usb dongle seems to have to program in different way of the bigger board. i see nrfutil in the tutorial. it's not possible to use the same toolchain? or i wrongly understand? thanks

caizelin, 2019-02-02:
> now i use the board with usbc female, i'm looking to shrink project board, and the usb dongle seems to have to program in different way of the bigger board. i see nrfutil in the tutorial. it's not possible to use the same toolchain? or i wrongly understand? thanks

The dongle does not contain a Debugger chip like nRF52840-MDK. So you can't use `pyocd` to program the dongle directly. 
However, the dongle is pre-installed with a bootloader. So you can use **nrfutil** or **nRF Connect for Desktop** to program the board. 
Development flow is the same as nRF52840-MDK except the programming step.