makerdiary/nrf52840-mdk #15: how to import sdk example?

stefanosky, 2019-02-02:
please can you explain how to use the others sdk examples, from Nordic.
like other example about nfc and ble.
in particular the import section and the board specification

thanks

caizelin, 2019-02-03:
Hi,
You can see [these](https://github.com/makerdiary/nrf52840-mdk/tree/master/examples/nrf5-sdk) examples. They are imported from nRF5 SDK. You can follow these steps to import a new one by yourself:
1. Copy an example from the nRF5 SDK to your workspace
2. Adapt the directories like this:
``` bash
.
├── armgcc
│   ├── Makefile
│   ├── _build
│   └── ble_app_blinky_gcc_nrf52.ld
├── config
│   └── sdk_config.h
├── hex
│   └── ble_app_blinky_nrf52840_mdk_s140.hex
└── main.c
```
3. Adapt the SDK base path in Makefile:
``` makefile
MDK_ROOT := ../../../..
SDK_ROOT := $(MDK_ROOT)/nrf_sdks/nRF5_SDK_15.2.0_9412b96
PROJ_DIR := ..
```
4. Specify the board flag in Makefile. Remember to add the include path for `nrf52840_mdk.h` and `custom_board.h`
``` makefile
CFLAGS += -DBOARD_CUSTOM
CFLAGS += -DNRF52840_MDK
... 
ASMFLAGS += -DBOARD_CUSTOM
ASMFLAGS += -DNRF52840_MDK
```
5. Set the `pyocd` command in Makefile:
``` makefile
.PHONY: flash flash_softdevice flash_all erase release

# Flash the program
flash: default
	@echo Flashing: $(OUTPUT_DIRECTORY)/nrf52840_xxaa.hex
	pyocd-flashtool -t nrf52 -se $(OUTPUT_DIRECTORY)/nrf52840_xxaa.hex

# Flash softdevice
flash_softdevice:
	@echo Flashing: s140_nrf52_6.1.0_softdevice.hex
	pyocd-flashtool -t nrf52 -se $(SDK_ROOT)/components/softdevice/s140/hex/s140_nrf52_6.1.0_softdevice.hex

# Merge application and softdevice, then flash
flash_all: default
	mergehex -m $(OUTPUT_DIRECTORY)/nrf52840_xxaa.hex $(SDK_ROOT)/components/softdevice/s140/hex/s140_nrf52_6.1.0_softdevice.hex -o $(OUTPUT_DIRECTORY)/nrf52840_xxaa_s140.hex
	@echo Flashing: $(OUTPUT_DIRECTORY)/nrf52840_xxaa_s140.hex
	pyocd-flashtool -t nrf52 -se $(OUTPUT_DIRECTORY)/nrf52840_xxaa_s140.hex

# Erase chip
erase:
	pyocd-flashtool -t nrf52 -ce

# Generate hex file
release:
	@echo Generating hex file...
	mergehex -m $(OUTPUT_DIRECTORY)/nrf52840_xxaa.hex $(SDK_ROOT)/components/softdevice/s140/hex/s140_nrf52_6.1.0_softdevice.hex -o $(PROJ_DIR)/hex/$(PROJECT_NAME).hex
```
6. If Log UART backend is used, remember to set `NRF_LOG_BACKEND_UART_TX_PIN` to `20` in `sdk_config.h`

That's it!

stefanosky, 2019-02-03:
thank you very much for this usefull tut :)
i will do it
ok, i'had problem with logging, i tried to insert logging in blink example, with no success.
so for logging it's only to put NRF_LOG_BACKEND_UART_TX_PIN 20 ? therarent other flags to activate?

caizelin, 2019-02-03:
> thank you very much for this usefull tut :)
> i will do it
> ok, i'had problem with logging, i tried to insert logging in blink example, with no success.
> so for logging it's only to put NRF_LOG_BACKEND_UART_TX_PIN 20 ? therarent other flags to activate?

Just for the examples with logging module enabled. Also seeing [this](https://github.com/makerdiary/nrf52840-mdk/tree/master/examples/nrf5-sdk/power_profiling_example) example and comparing with the blinky example, you will know how to enable a UART logging module.

stefanosky, 2019-02-03:
yess!! thanks  for the patience, you still help me very much, and i hope other people starting prototyping with your module.

now
uart at 115200 kbps produce strange counter jumping, the leds are ok, they blink, the counter jump

in the main of main.c

`
    while (true)
    {
	    conteggio++;
		if (NRF_LOG_PROCESS() == false){
			for (int i = 0; i < LEDS_NUMBER; i++)
		    {
				NRF_LOG_INFO("c %d", conteggio);
				NRF_LOG_INFO("blink al led %d .", i );
				nrf_delay_ms(100);
				bsp_board_led_invert(i);
		        nrf_delay_ms(1000);
		    }
			
		}
		nrf_delay_ms(1000);
		//NRF_LOG_FLUSH();
    }`

produce:

<info> app: blink al led 0 .
<info> app: c 22
<info> app: blink al led 1 .
<info> app: c 22
<info> app: blink al led 2 .
<info> app: c 28
<info> app: blink al led 0 .
<info> app: c 28
<info> app: blink al led 1 .
<info> app: c 28
<info> app: blink al led 2 .
<info> app: c 34
<info> app: blink al led 0 .
<info> app: c 34
<info> app: blink al led 1 .
<info> app: c 34
<info> app: blink al led 2 .
<info> app: c 40
<info> app: blink al led 0 .
<info> app: c 40
<info> app: blink al led 1 .
<info> app: c 40
<info> app: blink al led 2 .

stefanosky, 2019-02-03:
here a blink with log :)

#include <stdbool.h>
#include <stdint.h>
#include <string.h>

#include "nrf_delay.h"
#include "boards.h"

#include "nrf_log.h"
#include "nrf_log_ctrl.h"
#include "nrf_log_default_backends.h"

#define BOARD_NAME    "nRF52840-MDK"
#define STRING_EOL    "\r\n"
#define STRING_HEADER "-- Power Profiling example 2 --"STRING_EOL \
	"-- "BOARD_NAME " --"STRING_EOL	\
	"-- Compiled: "__DATE__ " "__TIME__ " --"STRING_EOL
#define STRING_HEADER2 "-- kiki --"

static void log_init(void)
{
    ret_code_t err_code = NRF_LOG_INIT(NULL);
    APP_ERROR_CHECK(err_code);
    NRF_LOG_DEFAULT_BACKENDS_INIT();
    nrf_delay_ms(100);
}

/**
 * \brief Main application function.
 *
 * Application entry point.
 *
 * \return program return value.
 */

int main(void)
{
	uint32_t conteggio=0;

	log_init();
	bsp_board_init(BSP_INIT_LEDS);
	
        nrf_delay_ms(100);

	NRF_LOG_INFO(STRING_HEADER);

	NRF_LOG_INFO("blink startedddd");
	nrf_delay_ms(100);
	NRF_LOG_FLUSH();
	
    /* Toggle LEDs. */
    // Enter main loop. 
	
    while (true)
    {
		
		conteggio++;
		NRF_LOG_INFO("c %d", conteggio);
		NRF_LOG_FLUSH();
		nrf_delay_ms(100);
		
			for (int i = 0; i < LEDS_NUMBER; i++)
		    {
				NRF_LOG_INFO("blink al led %d .", i );
				NRF_LOG_FLUSH();
				nrf_delay_ms(100);
				bsp_board_led_invert(i);
		        nrf_delay_ms(1000);
		    }
		
    }

}

caizelin, 2020-03-17:
I’m closing this issue because it has been inactive for a long time. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin