makerdiary/nrf52840-mdk #16: How to debug the program on the board

lukaszmargielewski, 2019-02-09:
I am able to flash and run the programs the board but can't find a tutorial or guide of how to debug them? Documentation claims DAPLINK supports / contains Jlink but f.e.x: Segger Embedded Studio can't see it (cannot connect to Jlink). I am able to log stuff but in many cases it is not enough. 
Can you help or am I missing something?

caizelin, 2019-02-09:
Hi,
DAPLink supports CMSIS-DAP. You can debug the board using Keil or IAR. Follow this discussion to get started: https://community.makerdiary.com/t/daplink-for-nrf5832-mdk/228/12

lukaszmargielewski, 2019-02-09:
@caizelin : I should have mentioned I am using Mac OS while Keil and IAR seem to support Windows only.

caizelin, 2019-02-10:
> @caizelin : I should have mentioned I am using Mac OS while Keil and IAR seem to support Windows only.

@lukaszmargielewski on macOS, you can try VS Code & [Cortex-Debug](https://marcelball.ca/projects/cortex-debug/). Cortex-Debug supports OpenOCD GDB Server and pyOCD GDB Server, so it is very convenient to debug code using VSC & Cortex-Debug. I will writing a post to get started with them soon.

caizelin, 2019-02-11:
@lukaszmargielewski Here is the tutorial: https://blog.makerdiary.com/debugging-nrf52840-mdk-with-vscode/

caizelin, 2020-03-17:
I’m closing this issue because it has been fixed. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin