makerdiary/nrf52840-mdk #17: how to configure wpantund ?

Yug11, 2019-03-25:
I am implementing the openthread on makerdiary's nRF52840-MDK using a Linux machine. 
It goes fine up to flashing the board. After that, the step where we have to configure wpantund, it constantly shows an error and it remains uninitialized. Help me resolve this issue .

caizelin, 2019-03-25:
> I am implementing the openthread on makerdiary's nRF52840-MDK using a Linux machine.
> It goes fine up to flashing the board. After that, the step where we have to configure wpantund, it constantly shows an error and it remains uninitialized. Help me resolve this issue .

@Yug11 You can follow this tutorial: https://wiki.makerdiary.com/nrf52840-mdk/openthread/#configure-wpantund

Yug11, 2019-03-25:
@caizelin 
I have been following the same tutorial but it is not working as at this step the terminal keeps on and nothing happens
[image]

sometimes it shows unable to open the socket and sometimes descriptor in a bad state

caizelin, 2019-03-25:
Make sure which serial port name you are using. It seems that `/dev/ttyACH0` does not exist.

Yug11, 2019-03-25:
but it does exist
[image]

caizelin, 2019-03-25:
> but it does exist
> [image]

Yes. But I saw you were using `ttyACH0` not `ttyACM0`. Did you try `ttyACM0` or `ttyACM1`?

Yug11, 2019-03-25:
ttyACM0

caizelin, 2019-03-25:
> ttyACM0

Then check the UART configuration before you compile openthread. In `openthread/examples/platforms/nrf52840/platform-config.h` file:
```
/**
 * @def UART_HWFC
 *
 * UART Hardware Flow Control.
 *
 * @brief Possible values:
 *         \ref NRF_UART_HWFC_ENABLED - HW Flow control enabled.
 *         \ref NRF_UART_HWFC_DISABLED - HW Flow control disabled.
 *
 */
#ifndef UART_HWFC
#define UART_HWFC NRF_UART_HWFC_DISABLED
#endif

/**
 * @def UART_PIN_TX
 *
 * UART TX Pin.
 *
 */
#ifndef UART_PIN_TX
#define UART_PIN_TX 20
#endif

/**
 * @def UART_PIN_RX
 *
 * UART RX Pin.
 *
 */
#ifndef UART_PIN_RX
#define UART_PIN_RX 19
#endif
```

Yug11, 2019-03-25:
[image]

I changed it to ttyACM0 then this happens

Yug11, 2019-03-25:
> Then check the UART configuration before you compile openthread. In `openthread/examples/platforms/nrf52840/platform-config.h` file:
>
> ```
> /**
>  * @def UART_HWFC
>  *
>  * UART Hardware Flow Control.
>  * [...]

Yes I did open this file and copied this configuration at the end

Yug11, 2019-03-25:
> Then check the UART configuration before you compile openthread. In `openthread/examples/platforms/nrf52840/platform-config.h` file:
>
> ```
> /**
>  * @def UART_HWFC
>  *
>  * UART Hardware Flow Control.
>  * [...]

@caizelin 
I need to add these lines at the end of the file or change the whole file??

caizelin, 2019-03-25:
@Yug11 Just define `UART_HWFC` to `NRF_UART_HWFC_DISABLED`, `UART_PIN_TX` to `20`, `UART_PIN_RX` to `19`

caizelin, 2019-03-25:
BTW, you can also try the pre-built ncp firmware [ot-ncp-ftd.hex](https://github.com/makerdiary/nrf52840-mdk/tree/master/firmware/openthread).

Yug11, 2019-03-25:
@caizelin I have tried both the methods  but none of it worked

[image]

Yug11, 2019-03-25:
once it went up to this stage and lagged, after that, I tried it again but the same error occurred

[image]