makerdiary/nrf52840-mdk #23: Noobie question: Non-USB power supply

steffensky, 2019-06-12:
Hi,

I want to connect the nRF52840-mdk to a power suppy other than the usb-c connector.
As I have fried other boards before I want to make sure I understand the documentation correctly.

Here's what I would do:
  - Connect the boards VIN to the something between +3.3V and +5.5V
  - Connect any of the boards GND to the corresponding GND

Is that correct?

Secondary question:
    What exactly are the exclamation marks in the pinout diagram indicating?
    
Thanks for the help

caizelin, 2019-06-12:
@steffensky you are right. Just connect the VIN to the external power with +3.3V~5.5V. And connect the GNDs together.
The exclamation marks mean that you should not operate the pins out of range.
Hope it's helpful for you!

steffensky, 2019-06-13:
That helps.
Thanks for the quick answer

closed 2019-06-13 by steffensky