makerdiary/nrf52840-mdk #25: I can't start with nrF52840 MDK board

jfrancoya, 2019-07-19:


jfrancoya, 2019-07-19:
> No description provided.
After updating the firmware, following the instructions given at:
https://wiki.makerdiary.com/nrf52840-mdk/daplink/#how-to-update-the-daplink-firmware

Anytime I connect it to the USB it appears the CRP DISABLED (D:) unit with the firmware.bin file, dated at 2/6/2009 with 64Kb. If I deleted again, it disappears, but if I remove and connect it again, it appears again the file with the same attributes in the same drive.
How can I continue...

caizelin, 2019-07-19:
> Anytime I connect it to the USB it appears the CRP DISABLED (D:) unit with the firmware.bin file, dated at 2/6/2009 with 64Kb. If I deleted again, it disappears, but if I remove and connect it [...]

Hi,
When ` CRP DISABLED (D:)` appears, remove `firmware.bin`, then drag and drop the new firmware to this disk. The latest firmware is located here: https://github.com/makerdiary/nrf52840-mdk/tree/master/firmware/daplink with the name `lpc11u35_nrf52840_mdk_if_crc_0253.bin`.

After the file copy operation completes, eject and replug the board into your computer normally (without holding down the RESET button). Your device now has the new firmware installed and mounts a drive named **DAPLINK**.

jonas-schievink, 2019-10-27:
I have the same issue and copying over a new firmware does nothing (I've tried all 3 files in the folder). It still shows up as CRP DISABLED after reattaching. Has the MDK been tested on Linux?

caizelin, 2019-10-28:
> I have the same issue and copying over a new firmware does nothing (I've tried all 3 files in the folder). It still shows up as CRP DISABLED after reattaching. Has the MDK been tested on Linux?

@jonas-schievink The MDK is tested on Ubuntu. It seems that the bin file is not updated correctly. Which distro did you use? Make sure the bin file is integrated or try it on a Windows PC.

jonas-schievink, 2019-10-28:
@caizelin I've tried this on an up-to-date Arch Linux system. First with a file manager (Krusader), then using `mount`/`rm`/`cp`/`umount` manually. Neither worked, but the firmware file did show up in the mounted directory correctly.

I then did it on a Windows VM, passing through the USB device, and it worked on the first try.

(this is probably really not your fault, but a bug in the NXP bootloader, which is in ROM, so it's not really fixable; documenting this issue would be good though)

caizelin, 2019-10-28:
@jonas-schievink Thanks for your feedback. On Linux/Mac, you can use command: `dd if={new_firmware.bin} of={firmware.bin} conv=notrunc`. It works fine. I hope it works for you.

jonas-schievink, 2019-10-28:
@caizelin Wow, that does actually work! Thanks a ton!

v01d, 2020-11-20:
I had problems trying to update firmware on nRF52832-MDK and the command suggested by @caizelin also worked. This should be documented somewhere.