makerdiary/nrf52840-mdk #26: Problem updating DAPLink firwmare for nrf52840-mdk

BensProxy, 2019-10-23:
Following instructions here:
https://wiki.makerdiary.com/nrf52840-mdk/daplink/

Running on MacOS Mojave Version 10.14.6 (18G95). 

I plug the board in holding reset and see the red light turn on, confirming it is in bootloader.

But I do not see the device at `/Volumes/CRP DISABLD`, so I'm not able to complete the next steps
```
$ sudo mount -u -w -o sync /Volumes/CRP DISABLD
$ cp -X <path to interface firmware file> /Volumes/CRP DISABLD/
```

BensProxy, 2019-10-23:
Oh wait, took a look at https://github.com/makerdiary/nrf52840-mdk/issues/14

> The dongle does not contain a Debugger chip like nRF52840-MDK. So you can't use pyocd to program the dongle directly.
However, the dongle is pre-installed with a bootloader. So you can use nrfutil or nRF Connect for Desktop to program the board.
Development flow is the same as nRF52840-MDK except the programming step.

caizelin, 2020-03-17:
I’m closing this issue because it has been inactive for a long time. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin