makerdiary/nrf52840-mdk #27: Unable to use User Button on Mbed

manchoz, 2019-10-26:
I'm not able to get the User Button working an any Mbed example which makes use of buttons nor in any new program.

Wherever I substitute `BUTTONx` in examples with `p32` or `P0_1` I'm not able to read or detect button status.

User Button works seamless on the other examples, such as [nrf5-sdk/ble_app_blinky](https://github.com/makerdiary/nrf52840-mdk/tree/master/examples/nrf5-sdk/ble_app_blinky).

Do you have any clue?

Thanks

manchoz, 2019-11-11:
At last, I was able to get the USR_BTN working.

The issue is caused by the `ITM` Trace support enabled by default for [`MCU_NRF52840`](https://github.com/manchoz/mbed-os/blob/manchoz_makerdiary-nrf52840-mdk/targets/targets.json#L7819) in MbedOS targets definitions.

On the MakerDiary nRF52840-MDK the USR_BTN is connected to the pin `P1.00` which is the default pin [`TRACEDATA0`](https://infocenter.nordicsemi.com/topic/ps_nrf52840/pin.html?cp=3_0_0_6_0_0#aqfn73) for the ITM trace port on the [nRF52840](https://infocenter.nordicsemi.com/topic/ps_nrf52840/dif.html?cp=3_0_0_3_7_4#trace) MCUs.

MbedOS [enables tracing](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/device/system_nrf52840.c#L201) at system initialization if the `ITM` support is enabled for the target, configuring the pre-defined pins for SWO tracing thus rendering impossibile using the `P1.00` button as `DigitalIn`.

In order to get `USR_BTN` working on MbedOS you need either to add `"device_has_remove": ["ITM"]` in [targets/targets.json](https://github.com/manchoz/mbed-os/blob/manchoz_makerdiary-nrf52840-mdk/targets/targets.json#L7921) definition or to add `"target.device_has_remove": ["ITM"]` in nRF52840-MDK overrides in `mbed_app.json` for your application.

Up-to-date MbedOS with support for nRF52840-MDK is available at [https://github.com/manchoz/mbed-os/tree/manchoz_makerdiary-nrf52840-mdk](https://github.com/manchoz/mbed-os/tree/manchoz_makerdiary-nrf52840-mdk).

I have added support for the nRF52840-MDK also in [mbed-os-tools](https://github.com/manchoz/mbed-os-tools/tree/makerdiary_nrf52840_mdk).

@caizelin I'm planning to issue the pull requests to upstream for both repos: please, let me know if you agree.

caizelin, 2019-11-11:
@manchoz Looks good to me! Thanks for your PRs.

closed 2020-01-07 by manchoz