makerdiary/nrf52840-mdk #28: arm mbed compile fails for TARGET=NRF52840_MDK

stefschu1510, 2019-11-04:
Trying to compile with the arm mbed CLI results in the following error:

_argument -m/--mcu: NRF52840_MDK is not a supported MCU. Supported MCUs are:
ARCH_BLE,                   ARCH_BLE_BOOT,
ARCH_BLE_OTA,               ARCH_GPRS,
..._

Is the documentation not up-to-date?
I'm using the latest mbed CLI, version 1.10.1

stefschu1510, 2019-11-05:
I was not aware that it is required to use the mbed-os fork maintained at makerdiary.
Hope there is a plan to include the NRF52840_MDK target to the official mbed-os.

closed 2019-11-05 by stefschu1510

manchoz, 2019-11-05:
@stefschu1510 you can find an upstream-updated port at [https://github.com/manchoz/mbed-os/tree/nRF5240_MDK](https://github.com/manchoz/mbed-os/tree/nRF5240_MDK).

I'm would like to PR to upstream but I'm blocked on #27.

manchoz, 2019-11-11:
Updated branch with fix for #27 available at https://github.com/manchoz/mbed-os-tools/tree/makerdiary_nrf52840_mdk