makerdiary/nrf52840-mdk #31: Build failed

wangyanjiong163, 2020-02-16:
Hi,

When I tried to follow the doc to build nRF5-SDK-for-Mesh app on my mac, it failed as below:

itch_client_nrf52840_xxAA_s140_6.1.0.dir/src/main.c.obj   -c ../examples/my_light_switch/client/src/main.c
../examples/my_light_switch/client/src/main.c: In function 'button_event_handler':
../examples/my_light_switch/client/src/main.c:228:18: error: 'BUTTON_GROVE_1' undeclared (first use in this function)
  228 |             case BUTTON_GROVE_1:
      |                  ^~~~~~~~~~~~~~
../examples/my_light_switch/client/src/main.c:228:18: note: each undeclared identifier is reported only once for each function it appears in
../examples/my_light_switch/client/src/main.c:237:18: error: 'BUTTON_GROVE_2' undeclared (first use in this function)
  237 |             case BUTTON_GROVE_2:
      |                  ^~~~~~~~~~~~~~
../examples/my_light_switch/client/src/main.c: In function 'buttons_init':
../examples/my_light_switch/client/src/main.c:285:10: error: 'BUTTON_GROVE_1' undeclared (first use in this function)
  285 |         {BUTTON_GROVE_1, BUTTONS_GROVE_ACTIVE_STATE, BUTTON_GROVE_PULL, button_event_handler},
      |          ^~~~~~~~~~~~~~
../examples/my_light_switch/client/src/main.c:285:26: error: 'BUTTONS_GROVE_ACTIVE_STATE' undeclared (first use in this function); did you mean 'BUTTONS_ACTIVE_STATE'?
  285 |         {BUTTON_GROVE_1, BUTTONS_GROVE_ACTIVE_STATE, BUTTON_GROVE_PULL, button_event_handler},
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                          BUTTONS_ACTIVE_STATE
../examples/my_light_switch/client/src/main.c:285:54: error: 'BUTTON_GROVE_PULL' undeclared (first use in this function); did you mean 'BUTTON_PULL'?
  285 |         {BUTTON_GROVE_1, BUTTONS_GROVE_ACTIVE_STATE, BUTTON_GROVE_PULL, button_event_handler},
      |                                                      ^~~~~~~~~~~~~~~~~
      |                                                      BUTTON_PULL
../examples/my_light_switch/client/src/main.c:286:10: error: 'BUTTON_GROVE_2' undeclared (first use in this function)
  286 |         {BUTTON_GROVE_2, BUTTONS_GROVE_ACTIVE_STATE, BUTTON_GROVE_PULL, button_event_handler}
      |          ^~~~~~~~~~~~~~
[10/132] Building C object external/micro-ecc/CMakeFiles/uECC_nrf52840_xxAA.dir/uECC.c.obj
ninja: build stopped: subcommand failed.

wangyanjiong163, 2020-02-19:
anyone could help on this?

caizelin, 2020-02-19:
> anyone could help on this?

Hi there, this example uses a Grove button for testing. You can modify the board config or just follow the tutorial here: https://wiki.makerdiary.com/grove-mesh-kit/bluetooth-mesh/

wangyanjiong163, 2020-02-20:
Hi, I followed 'Building the mesh stack and examples' tutorial in https://wiki.makerdiary.com/nrf52840-mdk/mesh/  site.