makerdiary/nrf52840-mdk #39: mdk board connected to usb port, but can't be found by pyocd-flashtool

wangyanjiong163, 2020-08-25:
Hi Experts,

I wanna to flash my mdk board, but my board can't be found by pyocd-flashtool. 

WARNING:pyocd.tools.flash_tool:pyocd-flashtool is deprecated; please use the new combined pyocd tool.
No available debug probes are connected

Maybe the bin is not the original one. Any way I can do to make sure the board could be seen by pyocd-flashtool?

Thanks.

xiongyihui, 2020-08-25:
Did you see a USB drive when the device was connected to your computer? If no USB drive is found, the device is detected by the computer.

wangyanjiong163, 2020-08-26:
What dev node should be seen under /dev after mdk connect to pc running ubuntu?  If such node not exist, what should I do to fix it?

xiongyihui, 2020-08-26:
You can use `lsusb` or `dmesg` to check if the device is detected on Ubuntu.

wangyanjiong163, 2020-08-27:
Yes，I have checked with lsusb. Nothing changed before and after the device connected. It seem not detected. 

Is that possible the fw in the mdk board not supporting fw update?

xiongyihui, 2020-08-28:
Did you use a USB hub with it? The nRF52840 MDK has a built-in USB hub. Too many levels of USB hubs may cause problems.

Or use an other computer to have a try.

wangyanjiong163, 2020-08-28:
Thanks Yihui. It is the USB hub issue.

closed 2020-08-28 by wangyanjiong163