makerdiary/nrf52840-mdk #46: Basic UART Communication not working on nrf52840-mdk

purvisb, 2020-12-31:
I cannot interface with the nrf52840-mdk via wpantund or CLI using openthread firmware.

I was following these [instructions](https://wiki.makerdiary.com/nrf52840-mdk/openthread/) originally with no luck. I went to the [openthread github](https://github.com/openthread/openthread/issues/6003) and asked for help, but still no luck. Please view that incident to see what troubleshoots I've already done.

I am able to open a CLI with the nrf52840-mdk using [this firmware](https://github.com/makerdiary/nrf52840-mdk/tree/master/firmware/openthread). But I can't open a CLI while using firmware I create with the make file. How was the firmware I linked to made? Are there some parameters I should be providing?

purvisb, 2021-01-07:
I resolved this issue by adding the parameter USB=1 and using /dev/ttyACM1 instead of /dev/ttyACM0. I would suggest adding this information to your guide.

closed 2021-01-07 by purvisb

songzh96, 2021-02-12:
Hi @purvisb 
I have the same problem as you, What is the compilation command you used？
I am using  `make -f examples/Makefile-nrf52840 USB=1`  

```
sudo /usr/local/sbin/wpantund -o Config:NCP:SocketPath /dev/ttyACM1 \
        -o Config:TUN:InterfaceName utun7 \
        -o Daemon:SyslogMask " -info"
```
this is my log output
```
wpantund[26321]: Starting wpantund 0.07.01 (Feb 11 2021 20:08:31) . . .
wpantund[26321]:        SOURCE_VERSION = 0.07.01
wpantund[26321]:        BUILD_VERSION = 0.07.01-2-g6993264
wpantund[26321]: Reading configuration from "/etc/wpantund.conf" . . .
wpantund[26321]: Configuration file "/etc/wpantund.conf" read.
wpantund[26321]: set-config-param: "Config:NCP:SocketPath" = "/dev/ttyACM1"
wpantund[26321]: set-config-param: "Config:TUN:InterfaceName" = "utun7"
wpantund[26321]: set-config-param: "Daemon:SyslogMask" = " -info"
wpantund[26321]: Ready. Using DBUS bus ":1.9674"
wpantund[26321]: Running as root without dropping privileges!
wpantund[26321]: SpinelNCPInstance-Protothreads.cpp:436: Requirement Failed (IS_EVENT_FROM_NCP(event) && GetInstance(this)->mInboundHeader == mLastHeader)
wpantund[26321]: Resetting and trying again... (retry 1)
wpantund[26321]: SpinelNCPInstance-Protothreads.cpp:426: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[26321]: Resetting and trying again... (retry 2)
wpantund[26321]: SpinelNCPInstance-Protothreads.cpp:426: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[26321]: Resetting and trying again... (retry 3)
wpantund[26321]: SpinelNCPInstance-Protothreads.cpp:426: Requirement Failed (event == EVENT_NCP_RESET)
wpantund[26321]: The NCP is misbehaving: Repeatedly unable to initialize NCP. Entering fault state.
wpantund[26321]: State change: "uninitialized" -> "uninitialized:fault"
```

purvisb, 2021-02-14:
@songzh96 That compilation command is the exact same as the one I used. Are you making sure to use the right symlink in /dev? There should be two (ttyACM0 & ttyACM1) in /dev. One is daplink, the other one is the NCP.

purvisb, 2021-02-14:
@songzh96 Oh I forgot that I already put that detail in my last comment that closed this ticket. You probably read that.

To make sure your using the right symlink atleast, you could use a udev rule. Create file: `/etc/udev/rules.d/99-usb-serial.rules`. And copy this text into it:
`ACTION=="add", SUBSYSTEM=="tty", ATTRS{idVendor}=="1915", ATTRS{idProduct}=="cafe", MODE:="0666", SYMLINK+="thr%n"`
 
Then restart. After that, when you plug in the nrf52 chip, a thr# symlink should populate in /dev. Where # is a integer.